manu7irl/klipper-DWC2-installer #1: Install?

ETE-Design, 2019-12-22:
So to understand it correct I just have to install Buster Lite, make an "git clone" and run your klipper-dwc2-installer.sh that's all?

manu7irl, 2019-12-23:
Hi, yes this pretty simple, clone it and run it.
I use it on raspbian lite for raspberry pi and armbian on orange pi.
If you have any questions let me know.

ETE-Design, 2020-01-03:
Have you tryed it on Buster Lite? Cause got a lot of errors, also have to add "Git" to Buster Lite, and nothing was in the *.sh after the clone, so have to copy / paste it also...

closed 2020-01-03 by ETE-Design

ETE-Design, 2020-01-03:
@manu7irl What to choose here for Buster Lite? The Debian just give access denied, and don't want it to install OctoPI...

=================================
      KLIPPER SERVICE INSTALL FOR SOC:
      ---------------------------------
      Please choose the install needed:
      (1)install-beaglebone.sh
      (2)install-centos.sh
      (3)install-debian.sh
      (4)install-octopi.sh
      (Q)uit
      ---------------------------------

manu7irl, 2020-01-03:
choose install-debian.sh these are not mine but the service install from
klipper, I just made a menu for that...

‫בתאריך יום ו׳, 3 בינו׳ 2020 ב-4:11 מאת ‪ETE-Design‬‏ <‪
notifications@github.com‬‏>:‬