manu7irl/klipper-DWC2-installer #7: Invoking of the Webinterface isn't possible

Peuqui, 2020-06-14:
...even with a clean installation described from here: https://klipper.info/klipper-+-dwc2-1/installing-klipper-with-dwc2 or with your scripts, also fresh and clean installed.
Then I tried to follow the "solutions" on several issues tracked from the original github sources and your responses there.
-> no luck.

manu7irl, 2020-06-14:
Did check for errors in your klippy-0.log?

Peuqui, 2020-06-14:
I got to manage to sort out this issue: Was a config error in regard of the serial connection: the [mcu] was doubled in the printer-0.cfg, so that klippy won't start.

Sorry for that!

closed 2020-06-14 by Peuqui

kaseiro98, 2020-11-06:
I am having this same error, and double MCU is not the problem here, what more could it be?