manu7irl/klipper-DWC2-installer #9: Proposal: Change the Klippy Log to be stored on the virtual SD card for DWC

lukeslaboratory, 2020-06-21:
Hey - I was just doing some updates to my config and I think it would be another neat step if the log file could be changed to be stored in the virtual SD card, where it could be seen by DWC under sys and you could interact with it there.  It beats every time needing to SSH into the machine to review and delete the klippy log when tracking down issues.

Thanks!

tripstar76, 2020-06-23:
Create a symlink....

ln -s /tmp/klippy.log ~/sdcard/sys/klippy.log

manu7irl, 2020-06-25:
good point I can add this, to the script.

manu7irl, 2020-07-03:
I added that on the last commit, please test if you see your log file in there...

manu7irl, 2020-07-03:
corrected a bad copy paste... now it should work..

closed 2020-07-03 by manu7irl