manu7irl/klipper-DWC2-installer #10: syntax error

TNT911, 2020-06-26:
Getting the following error:

###### Creating some folders...
./installer.sh: line 232: syntax error near unexpected token `else'

changedsoul, 2020-06-27:
Same error for me. Tried the old script to update klipper and dwc and was left with a broken printer. Came here to update the installer hoping there was a fix and see it was updated yesturday....yipie!!! only to get this error too and now Im bummed, lol

exxperto, 2020-06-29:
I'm not sure, but you can try change line 231 to this:

```
if [ -f $installLocation/printer.cfg ]; then report_status "Moving and renaming your printer.cfg to the correct folder... under $PRINTER_FOLDER/" && sleep 3 && mv $installLocation/printer.cfg $KlipperFarm/$PRINTER_FOLDER
```

julien-w, 2020-07-02:
Same error here...

manu7irl, 2020-07-03:
corrected that! fixed 3 commits ago, test and report, thanks!

closed 2020-07-03 by manu7irl