manu7irl/klipper-DWC2-installer #11: Web Interface keeps on reseting - no temp or axis controls available.

changedsoul, 2020-06-27:
I updated klipper today using this script and now my dwc web interface keeps rebooting. I see my temps for hotend and bed for a split second, then they disappear. over and over again. 
I checked the log and i didnt see any errors in there.

it reads the config file which has not changed since before the update, but this is whats after it.
`DWC2 starting at: http://0.0.0.0:4750
Extruder max_extrude_ratio=0.266081
Starting serial connect
Loaded MCU 'mcu' 80 commands (v0.8.0-617-g54166567-20200626_192601-octopi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'mcu' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB3,PB2,PB1 SERIAL_BAUD=250000 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PE0,PE1 BUS_PINS_twi=PD0,PD1 MCU=atmega2560 STATS_SUMSQ_BASE=256 CLOCK_FREQ=16000000
Sending MCU 'mcu' printer configuration...
Configured MCU 'mcu' (629 moves)
Starting heater checks for heater_bed
Starting heater checks for extruder
Stats 1193.7: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=1116 bytes_read=3368 bytes_retransmit=0 bytes_invalid=0 send_seq=113 receive_seq=106 retransmit_seq=0 srtt=0.006 rttvar=0.004 rto=0.025 ready_bytes=56 stalled_bytes=0 freq=16004067 heater_bed: target=0 temp=0.0 pwm=0.000  print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 sysload=0.08 cputime=9.518 memavail=751584
DWC2:225367.5 - 10.0.0.44 - GET - /rr_connect?password=reprap&time=2020-6-26T19%3A35%3A53
DWC2:25367.63 - 10.0.0.44 - GET - /rr_filelist?dir=0%3A%2Fmacros&first=0
Stats 1194.7: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=1183 bytes_read=3423 bytes_retransmit=0 bytes_invalid=0 send_seq=115 receive_seq=115 retransmit_seq=0 srtt=0.006 rttvar=0.005 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16004601 heater_bed: target=0 temp=0.0 pwm=0.000  print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=0.0 pwm=0.000 sysload=0.08 cputime=9.559 memavail=751552
DWC2:25368.15 - 10.0.0.44 - GET - /rr_config
DWC2:25368.61 - 10.0.0.44 - GET - /rr_config
Stats 1195.7: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=1189 bytes_read=3516 bytes_retransmit=0 bytes_invalid=0 send_seq=116 receive_seq=116 retransmit_seq=0 srtt=0.006 rttvar=0.004 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16004822 heater_bed: target=0 temp=29.9 pwm=0.000  print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=28.3 pwm=0.000 sysload=0.08 cputime=9.617 memavail=751552
Stats 1196.7: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=1195 bytes_read=3622 bytes_retransmit=0 bytes_invalid=0 send_seq=117 receive_seq=117 retransmit_seq=0 srtt=0.006 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16004777 heater_bed: target=0 temp=29.9 pwm=0.000  print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=29.2 pwm=0.000 sysload=0.08 cputime=9.652 memavail=751552
Stats 1197.7: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=1201 bytes_read=3728 bytes_retransmit=0 bytes_invalid=0 send_seq=118 receive_seq=118 retransmit_seq=0 srtt=0.005 rttvar=0.003 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=16004772 heater_bed: target=0 temp=30.1 pwm=0.000  print_time=0.001 buffer_time=0.000 print_stall=0 extruder: target=0 temp=29.2 pwm=0.000 sysload=0.07 cputime=9.687 memavail=751552
`

It looks like its getting the info, just for some reason the web page is freaking out.
I rebooted the PI and no good, still happens.

Any idea whats up?

tripstar76, 2020-06-27:
Try this

https://github.com/Stephan3/dwc2-for-klipper/commit/ffe00bf32baaf047b9e28b8fb41529fd65a5f9bc

changedsoul, 2020-06-27:
Thank you, that did the trick.

closed 2020-06-27 by changedsoul

apphead, 2020-09-01:
Did not my problem. Maybe not directly linked to a broken klipper release, see here for a fix:
https://github.com/Stephan3/dwc2-for-klipper/issues/101#issuecomment-679213417