manu7irl/klipper-DWC2-installer #12: Klipper not installing

sshaikh, 2020-07-07:
Fresh raspbian buster install.

I run the installer, use "pi" as my user for one instance.

Ended successfully with an empty printer-0.cfg and no services to start klipper.

An install log would be helpful as I have no idea what the script has done (or failed to do).

manu7irl, 2020-07-07:
The service is at klipper-0.service.
What's the issue?

sshaikh, 2020-07-07:
The issue is that the service doesn't exist and isn't enabled. The empty printer.cfg suggests to me that the installer didn't complete (even though it claimed to have).

manu7irl, 2020-07-07:
did you pull my latest commit?

‫בתאריך יום ג׳, 7 ביולי 2020 ב-20:42 מאת ‪sshaikh‬‏ <‪
notifications@github.com‬‏>:‬

manu7irl, 2020-07-07:
[image]
you should have at the end of the install a folder structure like this.

jornada812, 2020-09-13:
I am trying to follow your instructions but I cannot see the printer-0 folders. dwc is also not installed

asovetov, 2020-09-14:
Confirm. Latest raspbian, clear install. No folder printer-0, no default config, and no service klipper* in services or sysctl...
[image]
[image]

asovetov, 2020-09-14:
For me help manualy rerun ./scripts/install-octopi.sh