manu7irl/klipper-DWC2-installer #13: Stopping klipper-0 service...

damanbaird, 2020-07-17:
ERROR: Failed to stop klipper-0.service: Unit klipper-0.service not loaded
 done on fresh Raspberry Pi OS install 
Klipper installed fine, went to "flash your MCU" and got this.

manu7irl, 2020-07-17:
If you got first step successful, you should not have an error about a
missing klipper-0.service.
As the first step created it.
So eventually the first step was to blame.
Try from scratch the first step should let you stay in the script and also
prompt about a klipper-0 service that is running.

damanbaird, 2020-07-17:
Ok, let have another go at it, and ill keep you posted, thanks!

manu7irl, 2020-07-17:
I think, I spotted a little problem, I will fix it this saturday.
As a workaround create a printer.cfg file in your user folder.
/home/pi/printer.cfg
As my script tries to move over my new folder structure.

damanbaird, 2020-07-17:
I tried 3 more times before it finally said it was installed. Have one more issue though.

damanbaird, 2020-08-07:
I have no idea what i did but it has been working now.

closed 2020-08-07 by damanbaird

klHaribo, 2020-09-18:
I have the same error now, thinking to give up.
as an alternative, is it possible that the script uses the "/etc/init.d/klipper stop" command?
the first try, the script stops because ~/PrinterFarm/KlipperFarm does not exist...

another improvement: instead asking for the user, the script could ask the "whoami" command?

kaseiro98, 2020-11-06:
I am getting this same error, any idea why?

damanbaird, 2020-11-06:
Try from a fresh install again, think i tried 3 times before it went right for me.

kaseiro98, 2020-11-09:
> Try from a fresh install again, think i tried 3 times before it went right for me.

I didn't have enough memory on sd card I guess. I was trying initially on a 4gb sd card. Then swapped to a 16 GB card and this error was gone. thanks anyway