manu7irl/klipper-DWC2-installer #15: BUG: typo in installer for webcam

apphead, 2020-09-04:
https://github.com/manu7irl/klipper-DWC2-installer/blob/21ebac9067bad34b480988e4d2639ffcad987ab8/installer.sh#L755
it should say
	MJPGSTREAMER_HOME=$installLocation/mjpg-streamer/mjpg-streamer-experimental

Minor:
https://github.com/manu7irl/klipper-DWC2-installer/blob/21ebac9067bad34b480988e4d2639ffcad987ab8/install-multisession-octoprint.sh#L175
b is missing in bak directory: we**b**camDaemon_bak

manu7irl, 2020-09-04:
Hi thanks for noticing it... This script alone has been deprecated for
long...
As I run all the function from on main script.
I have still a lot to do to improve what we have from now especially the
octoprint installer part...

apphead, 2020-09-04:
`As I run all the function from on main script.`
What is the main script? If you are referring to klipper-DWC2-installer/installer.sh then see my first note, otherwise, I don't know what you mean by that.

As for my installation, I managed to get 2 instances of klipper & DWC up and running, but octoprint and webcam have failed so far. Good thing is that this installation might help me configure my _regular_ octopi & klipper & DWC combo - the latter never seems to work and webcam, well I safe that for a rainy day ;-)

Anyway, your script is really nice, my favourite being the muliple session thingie and haproxy. Maybe I will attempt that some day and study your scripts for that purpose.

**There is one last thing I urge you to do: Add a note in your readme that your project is very much a work in progress and not suited for the faint-hearted / less linux savy** 
But by all means, continue to develop it, I think it has great value and the potential to spare lots of headaches! Especially for beginners / intermediates.